pu2clr/SI4735 #2: Link to "AN332 REV 0.8 UNIVERSAL PROGRAMMING GUIDE"

mdanh2002, 2020-10-03:
Hi,

There are several references in the source code to "AN332 REV 0.8 UNIVERSAL PROGRAMMING GUIDE", especially regarding the SSB patch but I could not find such a document on silabs.com or on the web. The official AN332 programming guide https://www.silabs.com/documents/public/application-notes/AN332.pdf does not mention SSB.

Do you have the link to download this document? Thanks a lot.

pu2clr, 2020-10-03:
Sorry. I cannot find it on google search any more.
It was available before.

nvalw, 2020-10-03:
Hi!
All AN332 0.7, 0.8,1.0,1.2 in my archive  https://yadi.sk/d/t429W8OhrfSWmA
This is all that is on the home laptop, I will look on Monday, what is on the work laptop, there is a sufficient archive of versions of documents.
All the best to you!

> 
>  Sorry. I cannot find it on google search any more.
>It was available before.
>
>
>
>> Hi,
>>
>> There are several references in the source code to "AN332 REV 0.8
>> UNIVERSAL PROGRAMMING GUIDE", especially regarding the SSB patch but I
>> could not find such a document on silabs.com or on the web. The official
>> AN332 programming guide
>>  https://www.silabs.com/documents/public/application-notes/AN332.pdf does
>> not mention SSB.
>>
>> Do you have the link to download this document? Thanks a lot.
>>
>>
>
>
>
>QRZ:  http://www.qrz.com/db/PU2CLR
>Skype: ricardo_caratti
>Mobile:  +55 61 981442296

mdanh2002, 2020-10-03:
Thank you for the upload. Let me know if you manage to find the Universal Programming Guide.

nvalw, 2020-10-03:
I found in my stocks something similar to what you are looking for, but what? I'm not a programmer .... https://yadi.sk/i/UMO_unRrRnTpvg

nvalw, 2020-10-10:
New link - https://yadi.sk/i/KAfJNLo9r9_3xQ

pu2clr, 2020-10-19:
It is not a library issue.

closed 2020-10-18 by pu2clr