pu2clr/SI4735 #3: SI47XX_03_ALL_IN_ONE_NEW_INTERFACE.ino black screen / no status.

gbecki, 2020-10-13:
When running the example SI47XX_03_ALL_IN_ONE_NEW_INTERFACE.ino, the screen turns to a black right after splash text. 

**Expected behavior**
Status from the Si4735 device.

** Please, add the following information):**
 - Arduino IDE version: 1.8.9
 - Arduino: Nano
 - Display: OLED 0.91

**Additional context**
Running the example  SI47XX_02_ALL_IN_ONE_OLED.ino, the display shows the status just fine.

Thank you
PU5KNB - Guilherme.

pu2clr, 2020-10-13:
It is really curious. I have just tested this sketch with two OLED displays (0.9" and 1.3"). It is working fine here. 
Are you using the same hardware to check the both sketches (SI47XX_02_ALL_IN_ONE_OLED.ino and SI47XX_03_ALL_IN_ONE_NEW_INTERFACE.ino).

gbecki, 2020-10-13:
Yes exactly the same hardware.  I'll check if the new sketch activates the Si4735 properly.

gbecki, 2020-10-13:
The I2C adjustments works just fine.

Thanks.

closed 2020-10-13 by gbecki