pu2clr/SI4735 #5: Chip 4735 not tune

CloneTV, 2020-11-19:
Hi, I have some problems tuning to frequencies.
In the downloaded sketch SI4735_01_POC there is always one response from the console:

``
The Si473X I2C address is 0x11
You are tuned on 0.00MHz MONO [SNR:0dB Signal:0dBuV]
``
SI4735 connection diagram is typical, published in your repository. 
I checked the connections, everything is correct, I connected RESET to 7GPIO, GPIO8 used is setAudioMuteMcuPin().\

What is your opinion, what is the problem?

pu2clr, 2020-11-19:
When  the system does not start, most frequent problems are: 

1. Avoid using the computer connected to the mains during testing. The electrical grid can disturb the communication between the Arduino based board and the SI47XX device;
2. The RESET pin is not configured properly. Check the connection of the SI47XX pin 15 (RST) and the Arduino based board;
3. If the SI47XX pin 16 (SEN) is grounded, the I2C bus address must be 0x11, otherwise it must be 0x63 (the default I2C bus address is 0x11). Preferably, keep this pin grounded;
4. Check if the pins 17 (SCLK / SCL) and 18 (SDIO / SDA) of the SI47XX device are correctly connected to the Arduino board pins;
5. Check the pull-up resistors connected to the pins 17 (SCLK / SCL) and 18 (SDIO / SDA) of the SI47XX device;
6. Check the voltage on SI47XX pin 15 (RST).  It should be 3.3V. All digital pins of your Arduino must have 3.3v when in HIGH condition. If it is greater than 3.3V, probably you are using an 5V board; 
7. If you are using the board Arduino Pro Mini 3.3V (8MHz), be sure you selected the correct board on Tools menu, Processor: __"Atmega328P (3.3V, 8MHz)"__. By default, the Arduino IDE uses the 5V processor version; 
8. Do not try to power an ATmega328 Arduino Board designed to work with 5V and 16MHz with a 3.3V supply. That configuration will make your system unstable. See [ATmega328P Datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-7810-Automotive-Microcontrollers-ATmega328P_Datasheet.pdf);  
9. Do not try to power your Arduino based board designed to work with 3.3V with greater voltage. See the technical specifications 

See more on https://pu2clr.github.io/SI4735/#tips-to-build

clclon, 2020-12-03:
Thanks for the recommendations, it was a non-working quartz. 
On the diagram at address 
httpss://github.com/pu2clr/SI4735/blob/master/extras/images/basic_schematic_oled_buttons_eagle.png, the display power is incorrectly wired, VCC and GND are reversed.

pu2clr, 2020-12-03:
Thanks. The schematic was fixed. Thanks.

clclon, 2020-12-08:
If it's not difficult for you, tell me what methods change the modulation in the SW bands?  I would like to get AM, NFMmodulation and OFF SSB. 
There are many settings for SSB, but I did not find modulation change in the API description, only setAM, setFM, but this also changes the band, if I understand correctly ..

deskl0, 2021-03-06:
I'm suffering the same problem. The chip can be recognized by i2c scanner, however, it not responding to any commands. Measured xtal is functional, other i2c devices are working normally. This there any clues for that?

milestorm, 2021-03-18:
I have the same issue. Chip is recognised, but then it hangs. Does not even print "you are tuned...". But if I remove the reset pin cable, it resumes and prints the tuning to 0.00 MHz...

I double checked the wiring, it is connected to 3.3V ESP32. and connected to my laptop thru USB. The esp is also powered from USB

pu2clr, 2021-03-18:
Interrupt, SSB and NBFM are out of this issue topic. But I will comment about them here:

1) I have implemented interrupt support in the Library. However, I did not implement a true example showing how to use that. I intent of reviewing the strategy for dealing with interrupts. I think it can be done better out of Arduino Library.

2) About SSB, please check the function setSSB: https://pu2clr.github.io/SI4735/extras/apidoc/html/group__group17.html#ga7c48f2af416b1f89dbedbfbe9a637620

3) It is needed NBFM patch for SI4735-D60/SI4732-A10. I did not find one to check this mode. 

Regarding the tuning problem,   the real issue of this topic, check the comments in the  examples that use ESP32. 
There, you will find some advices to setup the ESP32 to deal with I2C.

milestorm, 2021-03-18:
I have tried to initialize i2C like from ESP32 example, but it is the same. I put oscilloscope on SDA pin and it seems, it is stucked on infinite loop, or the ESP still tries to send some signal, but chip does not respond.

https://i.imgur.com/rQdgUsR.jpg

It is possible, that my chip is busted? I really dont know  how to test it, if its busted...

deskl0, 2021-03-20:
well... For myself, I solved this issue by replacing the resonation capacitors.

closed 2021-05-07 by pu2clr