pu2clr/SI4735 #6: Feature request: Hamlib command support

oh3gai, 2021-02-01:
Hi!

Thanks for the awesome work with si47xx library. 

Could it be possible to implement an arduino ide example which uses hamlib rigctl protocol style commands and replies (RPRT 0 / 1) to set/get freq, mode, rssi and bw.

This would make possible to use this nice little receiver with nearly all ham digimode applications with simple netcat  tcp <> serial bridge redirector.

Thanks for the really great and well documented library!

br, Jarno OH3GAI

pu2clr, 2021-02-01:
Yes, I think so.
It is necessary to develop/implement  the  CAT protocol to the SI473X
device on the Arduino environment.
It can be a good project.

73.

oh3gai, 2021-02-02:
Hi! 

Thansk for answer. 

I don't have so much experience in coding, but I thought that the CAT is harder to implement. ie, GQRX sdr aplication has solved this by creating "dummy" remote server which share the same protocol and commands as rigctl(d) which allows different hamlib supported apps to control it.  ( https://gqrx.dk/doc/remote-control )

It could be possible to implement the same way but use the serial bridge between si4735 radio and pc.

Of course I agree that the CAT functionality is more sophisticated way to do it. I was just thinking which is harder without deep understanding of coding.

This si47xx library is great! 

Br, Jarno Oh3gai

closed 2021-05-07 by pu2clr