pu2clr/SI4735 #14: LW-MW-SW issue on ATS-20+

lelesrc, 2022-03-29:
I recently bought an ATS-20+ on aliexpress (https://www.aliexpress.com/item/1005001989352273.html).
The ATS-20+ model is a slightly modified version of the ATS-20. It has a dedicated USB-C port for recharge, a different knob and (I believe) an ESD protection on the antenna side. The board is labeled as "ATS-20+ SI4732 20220115".

I tested it briefly in SW band with the default firmware (it arrived with version 1.x.x installed). Reception was good, and I successfully tuned to a station 6000km away using a simple random wire antenna (not bad!)

Since I upgraded the firmware to SI473X_ALL_IN_ONE_OLED_RDS_CHINESE_V7 version, I receive only static on LW-MW-SW, while FM is working correctly.

I checked the track and the cap between the antenna and AM pin on the Si chip and all seems good.

I did a lot of research, but I couldn't find mentions of ATS-20+ firmware upgrades (nor failed, nor successful).

Do you have any idea of what could have gone wrong? Is there any test I can run to better understand the problem?

@pu2clr Thank you for your awesome project, and thank you in advance for your help!

pu2clr, 2022-03-29:
No idea. I do not know details about the ATS-20+. I will try check it. 
it is very unlikely that the firmware interferes with the sensitivity of the receiver.   
Do you have the original sketch (firmware)? 
Curious. Have you contacted the seller?

lelesrc, 2022-03-30:
Thank you for your kind answer. Unfortunately I don't have the original firmware - I will try contacting the seller and asking for it, but I don't have big expectations on this..

Anyway I can't rule out an hw problem - the radio worked great for a couple of days and the failure after the upgrade could be just a coincidence and not the real cause of the issue.

Here some pictures of the radio I took with my phone:

[image]

[image]

[image]

[image]

[image]

pu2clr, 2022-03-30:
I'm not sure, but I think the original firmware is based on this sketch: [https://github.com/pu2clr/SI4735/tree/master/examples/SI47XX_KITS/AliExpress/SI473X_ALL_IN_ONE_OLED_RDS_CHINESE_V1_5](https://github.com/pu2clr/SI4735/tree/master/examples/SI47XX_KITS/AliExpress/SI473X_ALL_IN_ONE_OLED_RDS_CHINESE_V1_5). 
Unfortunately, the manufacture does not show enough details about its firmware. 
I know it is based on my project and sketch examples.

lelesrc, 2022-03-30:
I uploaded 1_5 version, but it did not solve.

I remember the "original" display layout to be different (small numbers for frequency), so they could have used another (maybe older or modified) firmware. I remember for sure the start screen displayed your name though.

I contacted the vendor asking for the "original" firmware. If they provide it, I will test and let you know; in case it's a firmware problem, you can count on me for future tests on this model. I'm eager to support your project!

Thank you again for your support!

lelesrc, 2022-03-31:
Update:
I did a thorough check of the antenna circuit and everything is ok. I also attached my spectrum analyzer to the pin 8 (AMI) and the signal arrives loud and clear in the SW band. Seems that whatever wrong is happening, is happening within the si4732 chip or after. 

Still waiting for the original firmware from the vendor.

pu2clr, 2022-03-31:
I'm really curious about that. I would like to see that closer.  Well, let us wait for the vendor's response. If the problem persists, I will be at your disposal to try to identify the problem via some channel like WhatsApp, Google Meet or some tools to help us to fix it.

cmacia-mn, 2022-03-31:
Hi lelesrc,
I also received an ATS-20+ this week and upgraded its firmware to SI473X_ALL_IN_ONE_OLED_RDS_CHINESE_V8 (not V7) and my receiver is working fine in all bands after the upgrade.
I only noticed a small minor issue when the radio is in SSB mode and I will post a separate issue on it in case someone has a suggestion.
You  may want to give V8 a try.

lelesrc, 2022-04-02:
Final outcome: definitely, not a firmware problem, but a simple coincidence - the hw problem simply popped up right after my firmware update.

**It's safe to upgrade an ATS-20+ with the AliExpress/SI473X_ALL_IN_ONE_OLED_RDS_CHINESE_XX firmware, as tested by @cmacia-mn.**

The vendor finally sent me a firmware, but not the one installed on the radio when I received it (!). It's a firmware from R8CEH (nice ui though), and it doesn't solve the issue. FM works perfectly, while LW/HW/SW gives only static.

During my research I found this nice guy that has the very same problem on his ATS-20:
https://youtu.be/381139I2DCg?t=374
He has tons of testing equipment and way more experience than me and after some analysis he states that there's something wrong with the chip itself.

Thanks @cmacia-mn for your suggestion and thank you @pu2clr for your willingness to help a random user from the internet asking for support: you are awesome!

I obtained a partial (50%) refund from aliexpress, and probably I will order another ATS-20+ to get back in the game.

closed 2022-04-02 by lelesrc

pu2clr, 2022-04-15:
I am not sure, but probably the SI4732 AMI was damaged. That is the reason
I can think of.
Do you have experience with soldering SMD components?
I mean, are you able to replace the  SI4732 device?
Regards.
Ricardo.

cmacia-mn, 2022-04-15:
Hi Ricardo,

I had requested to join the Facebook group about 1 week ago but nobody has approved my request yet. Would you mind approving it as I would really like to read and learn about other people’s experiences with the SI4732 radios?

Thanks so much,

Carlos