pu2clr/SI4735 #18: patch_ssb_compressed: no such file or directory?

linbows, 2022-12-31:
i've been trying to update my generic ats-20 receiver but seems like you used to have patch_ssb_compressed.h inside the examples subfolders and now i only see one in root of the repo. i tried copying  patch_ssb_compressed.h to working dir but there's still an issue importing the file. Please see below for details:

working dir: ```examples\SI47XX_03_OLED_I2C\SI47XX_03_ALL_IN_ONE_NEW_INTERFACE_V20```
arduino ide error output: ``` C:\Users\Maks\Src\git\cloned\SI4735\examples\SI47XX_03_OLED_I2C\SI47XX_03_ALL_IN_ONE_NEW_INTERFACE_V20\SI47XX_03_ALL_IN_ONE_NEW_INTERFACE_V20.ino:81:10: fatal error: patch_ssb_compressed.h: No such file or directory
 #include <patch_ssb_compressed.h>    // Compressed SSB patch version (saving almost 1KB)
          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated. ```

pu2clr, 2023-01-16:
Solved. Please, check it out now. Thanks.

closed 2023-01-16 by pu2clr

linbows, 2023-01-22:
> Solved. Please, check it out now. Thanks. 

Thank you!