pu2clr/SI4735 #22: Kits AliExpress Encoder Wrong Direction

simplio, 2023-02-07:
**Describe the bug**
Direction of encoder wrong. rotating left frequency increase. rotating right frequency decrease. Is same bug for all other functions.

**To Reproduce**
Spin the encoder watch the display to see wrong direction.

**Expected behavior**
Rotating left frequency decrease, rotating right frequency increase. Should be same for all other functions.

**Screenshots**
If applicable, add screenshots or photo to help explain your problem.

** Please, add the following information):**
 - Arduino IDE version: 2.0.3
 - Arduino Board: Nano
 - Library version: 0.29.0
 - 

**Additional context**
Please fix it in the code. 
Thanks.

pu2clr, 2023-02-07:
It is not a problem with the Library code. Maybe the  manufacturer inverted the encoder connections.
I can not fix it in the Library code. If so, all the right encoder setup will work wrong. 
But if you have the sketch, you can fix it by change the lines as shown below: 

Look for the lines 

#define ENCODER_PIN_A 2
#define ENCODER_PIN_B 3

and change to

#define ENCODER_PIN_A 3
#define ENCODER_PIN_B 2

simplio, 2023-02-07:
Yes the ATS 20+ uses a better encoder which has reversed pins of the ATS 20. That change resolves the issue. Thanks!

closed 2023-02-07 by simplio