pu2clr/SI4735 #30: Proposal: CAT serial/USB interface

gpanico, 2023-09-22:
Hi guys, 

I was wondering if it would be possible to implement a CAT interface via USB or UART (in such case via a MAX232 adapter) to pilot the radio from a PC. 

I don't know if the Arduino while running the radio-code can also "listen" to the USB or to the UART and run some function, like on old microcomputers the UART I/O was triggering an interrupt that could be trapped. 
Or maybe in the main loop we still have some milliseconds to spare? 

Would that be technically feasible?

pu2clr, 2023-09-22:
Yes, it is technically feasible to implement a Computer-Aided Transceiver (CAT) interface via USB or UART to control the radio from a PC while using Arduino. The Arduino can run the radio code and also listen to the USB or UART for incoming commands.  Arduino has two pins (0 and 1) that can be used as UART.  You can have your Arduino code run in a main loop where it listens for incoming Serial commands and also manages the radio, for example with the SI4732. If you have a device that supports UART but not USB, you can indeed use a MAX232 adapter to interface it with the Arduino. The Arduino can then listen for incoming commands on its UART pins.

gpanico, 2023-09-22:
Thanks for your reply! 
In such case I'll start working on adding a MAX232 to the Arduino nano of my ATS-20+, and write just enough code to read the frequency and write it to the UART... 

My idea would be to re-implement a well known CAT protocol (e.g. Yaesu FT-857D?) so we can reuse tools like flrig without having to change much.

PU2CLR TNX, ES 73 de M0HHE, AR K